Sinapse-Energia/MIFARE-US #1: To implement I2C LCD Library

sinapseenergia, 2017-05-25:
IoT BaseBoard can integrate I2C 1.3 inch LCD
For this project this OLED should print Status Messages related to Card reading:

Reading Card
Registered in Server
Not Registered in server (Error)
System OK (Normal status)

The I2C address for the 1.3 OLED is 0x78
The reccomended library to use is:
U8gLib (But is deprecated)

U8g2 (Last Version):
https://github.com/olikraus/u8g2/wiki

soporteHW, 2017-08-14:
Created the Libraries folder and included the I2C LCD library. All files from 'https://github.com/olikraus/u8g2/wiki' have been included.

[image]

Now, it's mandatory to read all documentation about library integration. For test that HW is correct, I think I could implement a test code (calls functions to validate, print message, etc...)

soporteHW, 2017-08-16:
I have found a simpler library than the one previously described. Check here: https://github.com/4ilo/ssd1306-stm32HAL

Included in LCD Library the following files: fonts.c, fonts.h, ssd1306.c and ssd1306.h
This files include all functions to write message in LCD display through I2C protocol.

For example:

    ssd1306_Init();
	HAL_Delay(1000);
	ssd1306_Fill(Black);
	ssd1306_UpdateScreen();
	
	HAL_Delay(1000);
	
	ssd1306_SetCursor(1,2);
	ssd1306_WriteString("Not Registered",Font_7x10,White);
	ssd1306_SetCursor(1,19);
	ssd1306_WriteString("in server (Error)",Font_7x10,White);

With result:

[image]

francisjjp, 2017-09-25:
I suppose I can close this issue.

closed 2017-09-25 by francisjjp