Sinapse-Energia/MIFARE-US #2: To Implement RFID RC522 Functions

sinapseenergia, 2017-05-25:
Previously we were using PN532 NFC Module.
This module works using I2C (and SPI as well), and worked fine, but for new desing we would like to test another board (Cheaper and smaller)

The Board to integrate is RFID-RC522 Board

Check Below:
http://www.instructables.com/id/Arduino-RFID-Reader-MFRC522-Turorial/

There are many libraries for arduino and STM32, check some like this:
https://stm32f4-discovery.net/2014/07/library-23-read-rfid-tag-mfrc522-stm32f4xx-devices/
https://github.com/miguelbalboa/rfid

Module can work using SPI and I2C.

We want to use I2C mode (Be carefull, most of libraries implements SPI by default)

Access codes, and memory distribution with Sevilla Chip are in Redmine related project.

**Caution:

First of all, check if this module allows to read Mifare Model Sevilla´s Card**

sinapseenergia, 2017-05-25:
Fast Test done by Celia described here using a Pi

http://sinapse.bitnamiapp.com/redmine/boards/151/topics/687

sinapseenergia, 2017-05-25:
"MIFARE cards have ID to read at bloque 3, sector 16 con clave de lectura todo a FFFFFFFFFFFF"

More info here (Spanish):

http://sinapse.bitnamiapp.com/redmine/boards/151/topics/667

soporteHW, 2017-08-16:
Please, give me permissions in redmine to enter the project

ralcaide, 2017-08-16:
Done, you have already rights

soporteHW, 2017-08-18:
Some doubts about HW RFID connections:

Check the attached image:

[image]

- I2C iterface connection through SDA, SCL pins: OK
- VCC and GND pins: OK
- Where do IRQ and RST pins need to be connected?

soporteHW, 2017-08-21:
Comments about previuos commit:

- Library implements SPI protocol for MFRC522 target communications. It's needed to change defined functions to implement I2C protocol communication.
- Check the defined functions in Library  (https://stm32f4-discovery.net/2014/07/library-23-read-rfid-tag-mfrc522-stm32f4xx-devices/) in order to change it to implement I2C procotol communication.

sinapseenergia, 2017-08-22:
Check I2C library:

(This one seems OK, but no so much doccumentation:)

https://github.com/arozcan/MFRC522-I2C-Library

It seems , maybe we will had a HW modification to be done:

Read this doccument:

https://forum.arduino.cc/index.php?topic=442750.0

Check if our PCB is the same. It seems in some HW revisions,

_" Depending on signals on pins 32 - EA, and 1 - I2C it is possible to set the module in desired mode.

If we want I2C what would be better in many cases, than we need to desolder the MRFC522 IC, because I2C enable line is grounded (set LOW) under the chip, and we need to reflow the  IC after cutting."_

Anyway, check with IC datasheet, compare with our pre-made board, check our board signals, and confirm that we need to do at HW level.

soporteHW, 2017-08-24:
No progress on RFID integration. I've tried to implement I2C protocol as used in other projects to write and read RTC device for example. Results are not successful!

Please, check the following datashet for MFRC522 IC: https://www.nxp.com/docs/en/data-sheet/MFRC522.pdf

Pay special attention to _Section 8.1.4 I2C-bus interface_

@sinapseenergia @francisjjp @ralcaide Could you help me?

sinapseenergia, 2017-08-25:
First thing that you hace to check if it I2C mode is enable.

As described in datasheet:

[image]

I2C Pin have to be to "1"

This pin, if you check page 6, correspond to PIN 1

[image]

Check this as first steep. If I2C is not enable, nothing is going to work

sinapseenergia, 2017-08-25:
As described in previous messages , IC has 3 working modes, selectable by EA nad I2C pins.

We ar working with an assembly PCB wich puts I2C to GND and EA to VCC this is SPI Mode

SPI interface is not available into IoT BaseBoard (To fix in future revisions)

So options are:

I2C and UART modes

I2C: Implies to desolder IC an cut 2 tracks
UART: Implies to cut EA track 

Both interfaces are available into BaseBoard so we are going to work with UART mode

**Note: It´s neccesary to cut EA track for this mode, and put the pin to GND (reccomendable)**

There are UART Library available here:

https://github.com/mfdogalindo/MFRC522-UART (Python)

https://github.com/zodier/MFRC522-UART-Arduino (Cpp)

ralcaide, 2017-08-25:
Only for info:

This project is  not ready to use .cpp libraries. To do that is necessary to develop a C-CPP interface, it is possible but is not totally easy. It is "better" to use a .c library or to develop a .c library only with the necessary commands. 

It is better only if it is fast to do, if not, we will develop the C-CPP interface. 

@soporteHW , are you able to find a C library or to develop a the C functions that we need based on the CPP library. If not, please tell us in order to developed an interface for the CPP library

ralcaide, 2017-09-04:
How to use a C++ library in a C based code: http://sinapse.bitnamiapp.com/redmine/projects/firmware-global/wiki/How_to_use_C++_libraries_in_C_based_code

francisjjp, 2017-09-06:
I have tested with UART5 and I am supossing that EA track is right cut in order to get UART working mode.

```

MX_USART5_UART_Init();

  /// test NFC
uint8_t addressRFID = (0x37|0x80); //if LSB is sent first in UART (0x37 is for getting version of NFC)
  //uint8_t addressRFID = (0xED); //if MSB is sent first in UART (this is 0x37|0x80 at oppsite)
  uint8_t dataRFID =0;
  HAL_UART_Transmit(&huart5,&addressRFID,1,1000);
  HAL_UART_Receive(&huart5,&dataRFID,1,1000);
```
We should see some data on dataRFID. One value bigger that 0x90. But I haven't seen yet.

It is needed to check if EA track is right cut and another track is not cut. (I suspect that pin 3 is cut by mistake).

francisjjp, 2017-09-07:
Only there is to test Seville Card to check if it is detected. There is the basic example structure in main.c

```
 if (0) // To check with real card.
   {

	   uint8_t CardID[5];
	   uint8_t MyID[5]=  { 0x43, 0xdc, 0x52, 0xb6, 0x7b };// My card on my keys

	   while (1) /// example with infinite loop
	   {
		  /// If any card detected

		   if (TM_MFRC522_Check(CardID) == MI_OK)
		   { // CardID is valid

		      /// Check if this is my card

			   if (TM_MFRC522_Compare(CardID,MyID)==MI_OK)
			   {
				   /// Detected my card.

			   }
			   else
			   {
				   // It is not my card.

			   }
		   }

		   else
		   {
			   // Card not detected.

		   }

	   }

   }
```

francisjjp, 2017-09-11:
It is needed to get data from one determined block. This is made by doing one authentication with the generic ID from card, and then read determined sector and block with specific key. In general 0xFFFFFFFFFFFF.

soporteHW, 2017-09-14:
Trying to read the MIFARE card.
We use the following function:

TM_MFRC522_Status_t TM_MFRC522_Auth(uint8_t authMode, uint8_t BlockAddr, uint8_t* Sectorkey, uint8_t* serNum) {
	TM_MFRC522_Status_t status;
	uint16_t recvBits;
	uint8_t i;
	uint8_t buff[12];

	//Verify the command block address + sector + password + card serial number
	buff[0] = authMode;
	buff[1] = BlockAddr;
	for (i = 0; i < 6; i++) {
		buff[i+2] = *(Sectorkey+i);
	}
	for (i=0; i<4; i++) {
		buff[i+8] = *(serNum+i);
	}
	status = TM_MFRC522_ToCard(PCD_AUTHENT, buff, 12, buff, &recvBits);

	if ((status != MI_OK) || (!(readRegister(MFRC522_REG_STATUS2) & 0x08))) {
		status = MI_ERR;
	}

	return status;
}

with the following params:
- authMode = 0x60 -> authentication key A
- BlockAddr = (16*4)+3
- Sectorkey = FFFFFFFFFFFF
- serNum = idCard -> Domingo's identifier

TM_MFRC522_ToCard returns "status = OK" but then when we try to read MFRC522_REG_STATUS2 it returns "status = ERR"

soporteHW, 2017-09-15:
It's necessary to check the original Celia program, in particular the following file library -> "/home/pi/Desktop/libnfc-1.7.1/utils/quick_start_example2 r a mycard2.mfd"

Check if the frame to authenticate the US card is in correct format.

soporteHW, 2017-09-18:
The authentication process is not work fine. We are trying to authenticate any sector (0, 7, 8, 63...) but the results are NOK. Something is wrong and we dont't identificate why.

soporteHW, 2017-09-18:
This issue was post in the following discussion:

- https://github.com/mxgxw/MFRC522-python/issues/24
- https://stm32f4-discovery.net/2014/07/library-23-read-rfid-tag-mfrc522-stm32f4xx-devices/

francisjjp, 2017-09-20:
This comment has been posted to Particle forum, STM community and STM32duino forum.

**Hello.

First. I have tested one python script (READ.py) with one raspberry and the MFRC522 works fine. It detects UID, it does SelectTag process as ok, it does Authentication process and It reads sector. All work ok

https://github.com/mxgxw/MFRC522-python

I have started to have problem when I have migrated this code to STM32F030CC microcontroller. Migrating code was easy. Only SPI process was a little more complicated. Finally I did two versions: one version by simulating MISO, MOSI, SCK and CS with GPIO and another one using SPI1. MISO, MOSI, SCK with SPI1 and CS manually with GPIO.

With these two versions I can read any register from MFRC522, I can detect card and get UID serial number, problem is that output for SelectTAG process is MI_NOTAGERR instead MI_OK, and process for Authentication process is MI_ERR instead MI_OK. First process for detect card requires to send to FIFO_DATA_REGISTER into MFRC522 only 1 byte, for getting UID serial requires to send to FIFO_DATA_REGISTER 2 bytes. For rest of process the number of bytes are more than 2 bytes (9 or more). 

Someone has worked with this, and knows the problem?

Best regards!

Francis.**

francisjjp, 2017-09-20:
To watch this issue that happens in SPI library loaded in script python that works ok:

https://github.com/lthiery/SPI-Py/commit/2c6cf5c8caa6157f84617de5e31af8e48ddbd8e7

```
Normally CS is asserted between messages then deasserted after the final
one. Setting cs_change to 1 deasserts CS between messages, but if set on
the final message it causes CS to remain asserted until the next transfer.
Counterintuitively, in the single message case, setting cs_change to 1
prevents CS from changing, while setting it to 0 allows it to change after
each message/transfer.

Some device, like the MFRC522 rely on a change of CS for synchronisation,
and you wouldn't want to leave multiple devices selected simultaneously,
so cs_change should be zero.

This issue was previously masked by the use of the BCM2835 hardware chip
selects, which don't allow CS to remain asserted between transfers.`
```

francisjjp, 2017-09-23:
Regarding this. There is one issue with RTC-Init. RFID seems no to work when RTC intialized. To watch clock.

soporteHW, 2017-09-24:
Solved problem with _**MX_RTC_Init()**_. The problem was in **_void SystemClock_config(void)_** function, it was neccessary to include some code sentences to set internal RTC.

francisjjp, 2017-09-25:
I suppose I can close.

luna-lovecraft, 2021-03-16:
tengo un detalle, estoy usando los ejemplos que vienen en la biblioteca del mfrc522, estoy usando el de escritura de personal, pero me sale el siguiente error, ya lei todo el codigo y no tengo la opcion o manera de meter informacion personalizada a la tarjeta, me pueden ayudar soy nuevo con este modulo
[image]

[image]

lasuridze, 2021-11-06:
hello can i use key A and Key B both for more security ?