Sinapse-Energia/MIFARE-US #3: To design & define the flow diagram of the main program

ralcaide, 2017-05-31:
It is necessary to define the flow diagram of the firmware. The main should be based in the python program available in the code

ralcaide, 2017-05-31:
Old Main program here: https://github.com/Sinapse-Energia/MIFARE-FreeRTOS/tree/master/third-party-old

soporteHW, 2017-07-28:
At date 27/07/2017

[13:00 -14:00] Reading information about StarUML application.

soporteHW, 2017-07-28:
First version flow diagram. I know it's a very iregular flow diagram. Sorry I don't manage the tool well.
Check and fix errors in order to review the flow diagram

[image]

soporteHW, 2017-08-14:
New flow diagram version uploaded.
[image]

Please check it!

sinapseenergia, 2017-08-14:
Some comments
:startup
1. Inizialization periphericals and Setup Context (update and load persistent variables and setup initial values)

Proccess 
Main  (Wait for IRQ and maintain status (like NTP time)

IRQ - RFID message implies (as consequence, different functions)

Read RFID (IRQ routine)
If Ok
Beep and update LCD
Mount http frame
Send to server
Wait for response (unti ltimeout)
Beeps and update LCD with results
If NOK 
Beeps and update LCD

Note: consider a proper attention to the case of new IRQ before sending.  Process starts again (with new lecture). Se have to consider always last reading

sinapseenergia, 2017-08-14:
Recondider program flow with comments

ralcaide, 2017-08-16:
The workflow does not explain very well how the program should work. The WF should display the iterations and the logic in global view in order to ease the understanding.

Some concrete remarks:

- Please represent the HW part, this part is the responsible of the IRQ generations. The Main does not generate IRQs
- A process can not start in the SB. There are only to points where a process can start: HW or Main. For example, the process 1. LCD_Status is a process that should start in the Main, in each iteration, the Main code will call a SB function that will display the Status in the LCD
- Process 2. What is that exactly? Is a sleep? 
- Process 6. This is call only if the process 5. was OK, isn't?. Also, all the sub-processes (calls triggered from other process) should be displayed inside a process. 

Please, see here some WFs more realistic https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Flow-diagrams, these flows maybe could help you to understand how to do the WFs in a better way

@soporteHW, take into account the @sinapseenergia comments a my comments and upload a new version.

soporteHW, 2017-08-16:
New WF version:
[image]

ralcaide, 2017-08-17:
Comments - To improve:

- Put the HW Lifeline in the last position. The FW before the HW
- 3: Where is this routine developed? Main? Other file?
- 4: Update context. Where are the vars saved? The management of the Context normally is SB because works with memory (EEPROM, RAM?)
- 5: What happens if NOK
- 6: This process is invoked in each iteration of the Main. If it is invoked only when 4 was OK, then the process should start in inside the first process
- 6: Get_HTTP_connection makes all related with the HTTP, isn't? Create message, send and receive? All these process are not in background?
- 8: This function call a SB function in order to send data through ETH
-9 : This function call a SB function in order to get data through ETH. Or is an interruption?
- 10: The end of the 6 is only by timeout? I don't think so, it should be also an out related with the response of the server
- 10: The response of the server is not analyzed?
-

ralcaide, 2017-08-18:
New file uploaded: https://github.com/Sinapse-Energia/MIFARE-US/commit/22b9f01e21da5f745c447a7c505c4aba5c99bcef

ralcaide, 2017-08-18:
Now, WF is much better. Anyway, the if_else statements should be improved. Please take into account the following comments:
- The process 1 (Init Periph) should end after the response of the Context, process 4. The first white rectangle should finish after 4.
- If flag = 1: Put it over the life line. Display also the else statement and the end of this statement with a "END if flag = 1"
- If RFID OK: Same remarks as for If flag = 1

soporteHW, 2017-08-21:
New file uploaded: https://github.com/Sinapse-Energia/MIFARE-US/commit/5625dabdb1949f8f00f59ceab346fa3cf4e61104

sinapseenergia, 2017-09-20:
**Important:**

Initial all periphericals fucntion:

As ST reccomends, all not used GPIOS, have to be definet as **OUTPUT** and inizializated to **LOW** level.
Low level Set have to be done in this function.

EMI/EMC ST Reccomendations uploaded for reference

[en.CD00004479.pdf](https://github.com/Sinapse-Energia/MIFARE-US/files/1317323/en.CD00004479.pdf)