Sinapse-Energia/MIFARE-US #4: To define Program Context

sinapseenergia, 2017-07-25:
Define context and memory types / allocation for this program

soporteHW, 2017-08-21:
To define the context, each device must store the following variables:

- IP
- ClientType
- Time
- Building
- Classroom
- ClassroomType
- User
- Password

@sinapseenergia Are those variables enough?  Does necessary to define others variables?

ralcaide, 2017-08-22:
@soporteHW , at first these variables are enough, if it is necessary to add more variables could be added without big problems.
Anyway, this issue is not done, you should define:
- Where (NVM) will be these variables saved?
- Type and size of each variable
- How will be charged into RAM / updated / saved in the NVM

See https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Context-Management for info about how you should finish this issue. 
Everything should be wrote in the WIKI

soporteHW, 2017-08-23:
Some doubts about variables type:

- IP server: ej. 10.10.10.254:80 -> OK, indicates IP server address
- IP: ej. 10.10.10.5 -> OK, indicates device IP address
- ClientType: It is a number? (1,2,3,4 according original project documentation)
- Time: ej. 23/08/2017 , 13:38:00 -> OK, indicates server schedule time
- Building: It is a number or a string?
- ClassroomType: it's a number -> OK
- Classroom: it's a number -> OK
- User & pass: according original project documentation, this variables was replaced. Sure?

ralcaide, 2017-08-24:
@soporteHW , I'm not sure about your question. We should read the doc and maybe ask to @sinapseenergia,
Anyway you can define the:
- ClientType as a string with 5 chars
- Building as a string with 20 chars
- Time; Should save a timestamp, not the date as a human readable format. Timestamp is a long number

sinapseenergia, 2017-08-25:
@soporteHW I don´t understand well your doubts regarding this issue.

For context definition, we should have every field, even if it is not in use. (I don´t know if if it what you mean with this comment:)

_"User & pass: according original project documentation, this variables was replaced. Sure?"_

Regarding the other fields, as @ralcaide indicates.

If you have doubst about messaging, in thye old project doccumentation, you can access to python code of previous device. In this code you can read easily how the frames is mounted and how the field are considered.

As main indication, for this application we have no memory limitations, so ,in general , add every variable which could be considered (Now or in the future). Even if now is not in use

sinapseenergia, 2017-08-28:
Regarding las comments about this , take into consideration:

2 variables for time:

Server_Time : It comes from the (NTP) server , this is the reference for updating local time when proceeds
Local_Time: Local time for device, it increases using the own RTC on device, and it´s updated with NTP time (Server) when proceed

@ralcaide in other order of things

For this application (Sevilla), the next variables are numbers:
Building: It is a number 
ClassroomType: it's a number -> OK
Classroom: it's a number -> OK

But, there are just information provided by server an to be represented into LCD

Maybe fit their type to "number" , could limitate future aplications were for instance we want to name a building as string.

Could be a good approaching to define all server data as strings.

@ralcaide  Please comment about this

francisjjp, 2017-09-08:
Regarding procedure of changing variable in HiLink webserver and context.

Only the momment of checking HiLink webserver changes is in restarting, @ralcaide is it?.

I.E: The installer want to change one IP for another one. He changes it and then he restarts the device. Then, at beginning of restarting, we do a Get_HK_Config(), and we see if some parameter has changed if comparing with context, then we save the changed value to context.

Is it this way for procedure?

On the other hand. What's happens if installer has a mistakes and we lost HK wifi?. Maybe is interesting at beginning to use http or xml frame for autodiscovering too. I mean. If we lost the IP, we connect the device to one router and with wireshark or another HTTP sniffer program, because of this sending of frame at beginning, we find out the IP of device.