Sinapse-Energia/MIFARE-US #5: To include HTTP Client

sinapseenergia, 2017-07-25:
Include HTTP Clietn designed by Sergii and Seka

sinapseenergia, 2017-07-27:
1. Mount HTTP Messages
2. Send trough HTTP Client

soporteHW, 2017-08-22:
@sinapseenergia @ralcaide Where can I find the http client code designed by Seka?

ralcaide, 2017-08-22:
In the Bootloader project,  I think is the this file: https://github.com/Sinapse-Energia/bootloader-stm32/blob/master/Src/Bootloader.c

Anyway, I don't see any http client packed. It seems they are using directly a socket over WIFI or GPRS in order to ask / receive HTTP. 
Please study the code and try to understand how to integrate it in our needs

ralcaide, 2017-08-24:
The code in the Bootloader is designed in the following way:
- Sockect_bank.c is the wrapper of the different communications options: GPRS & WIFI (232). In this file are the functions to connect, send and receive data using GPRS or WIFI. These functions are the "HTTP Client"
- M95lite.c : Functions to use the M95 modem as a Link layer (OSI Model)
- Wifi232.c Funtions to use the device WIFI232 as a Link layer. This device has already a lot of logic implemented and basically there is a function to send the http request to the UART where the Wifi 232 module is connected

Only the socket part is useful but only to get ideas because the socket part should be developed as a SB functions for this project but is missing the most important part: The Hilink driver. If we integrate the Hilink driver in #17.

This issue is really not useful

francisjjp, 2017-09-11:
It is needed for including in firmware to test NTP post and HTTP post with Seville server.

@sinapseenergia (Alvaro) told us that there is one pc in Sinapse connected to Seville LAN. Then it is possible to emulate by terminal device HTTP messages using one specific IP.  With this task , we will sure that HTTP messages are sent right.

For receiving, we will pick up the response from this PC connected to Seville LAN and then will put into firmware code in order to emulate the different server answers.

francisjjp, 2017-09-13:
The connection with Seville server uses HTTPS (HTTP + TLS) but **disable_ssl_certificate_validation=True**. What happens?. Does exist encryption? Is it only some string handshakes?

ralcaide, 2017-09-13:
Is not validating the authenticity of the certificate. It is using a certificate that will be valid always (It not a very secure mechanism)

soporteHW, 2017-09-20:
As shown in older version, the device interaction with US server performs the following process:

**1)** At first, MIFARE device sends a GET request to server in order to locate and indentify the device. The HTTP message payload is (an example):
 ```
<horde>
<aulas>
<IP>10.5.6.24</IP></aulas></horfeus>
```
The server response is:
```
<?xml version="1.0" encoding="utf-8"?>
<horfeus>
<aulas>
<IP>10.5.6.24</IP>
<resultado>
<dispositivo>102</dispositivo>
<lista>
<aula>
<codigo>RM120022</codigo>
<edificio>43</edificio>
<tipo>4</tipo>
<nombre>AULA 2.2 FACULTAD DE MATEMATICAS</nombre></aula></lista></resultado></aulas></horfeus>
```
From this string, we are only interested in <IP>,<codigo>,<edificio><tipo> fields to save them in Context.

**_NOTE: THIS COMMENT IS INCOMPLETE, UPDATE LATER._**

soporteHW, 2017-09-21:
Included the following structures in "_southbound_mifare.h_" file:

**_typedef struct Start_TAGS_** -> define all initial possible tags included in payload message. (e.g.: `<horfeus>`)

_**typedef struct End_TAGS**_ -> define all end possible tags included in payload message. (e.g.: `</horfeus>`)

Implemented the following functions in "_southboudn_mifare.c_" file:

void **_FillTags_**(); =>This function fills all defined tags (Start and End) with the correct names (e.g. horfeus, aula, IP, etc)

void **_encode_GET_msg_**(char *GET_msg); => This function encodes a string to use it in GET request message.

soporteHW, 2017-09-27:
Implemented and included in "southbound_mifare" file the following function:

_uint8_t **decodeXML**(char *bufferdecode);_
This function receive as parameter the server response to a HTTP request. For MIFARE application, I have only considered the next cases:

1) Server response to synchronization/location message (HTTP GET request). Here, we are only interested in tags "codigo", "edificio","tipo" in order to save his content in Context and Flash.

2) Server response to register card message (HTTP GET request). Here note: to register a valid card, finally we'll use GET method because Domingo doesn't interpret if the method is GET or POST, simply he does $request. So, to keep the GET message which it has been validate, we'll send a GET message to register a valid card)

soporteHW, 2017-10-02:
Included some functions:

- char ***_Build_HTTP_msg_**(HTTP_METHOD method, char *Payload);

This function builds the correct HTTP message (HEADERS + PAYLOAD) as appropiate. Returns the built message.

- uint8_t **_HTTP_request_**(char *HTTPbuffer);

This function sends an HTTP request to a server. Returns 1 if server response is OK, 0 otherwise.