Sinapse-Energia/MIFARE-US #6: [SB] Wifi PCB driver

sinapseenergia, 2017-07-25:
This BaseBoard uses HiLink Wifi.

TCP Conections/ sending subfunctions have an higguer jerarquy

Integrate accordly the defined structure the use and management of this Wifi PCB

sinapseenergia, 2017-07-27:
It´s neccesary to include in this issue the configuration functionalities

1. Read and Set HiLik Wifi parameters 
2. Send and Receive Uart Data from Wifi

This means 2 different wraper sub-functions to obtain all device functionalities

ralcaide, 2017-08-24:
@soporteHW , how is going this integration?. This is the first step in order to implement #17. 
Please, add the code necessary to work with HiLink device and test it from the main.c. Then, the SB functions will be designed in order to use the HiLink device as a Link layer

sinapseenergia, 2017-08-25:
First thing that you hace to check if it I2C mode is enable.

As described in datasheet:

[image]

I2C Pin have to be to "1"

This pin, if you check page 6, correspond to PIN 1

[image]

Check this as first steep. If I2C is not enable, nothing is going to work

soporteHW, 2017-09-04:
Implemented the following functions included in "southbound_mifare" files:

- uint8_t **_sendingATCommands_**(UART_HandleTypeDef *phuart1, uint32_t timeoutTx, uint32_t 
                                  timeoutRx, uint32_t numberOfReceivedBytes, unsigned char *messageTX, unsigned 
                                  char *messageRX);
- void **_CleanBufferReception_**(void);

soporteHW, 2017-09-05:
Moved **_sendingATCommands_**(*) function from _southbound_mifare_ file to _southbound_generic_ file.
Changes in  https://github.com/Sinapse-Energia/MIFARE-US/commit/499e0853622cacccb65443efc3f5df0c7ad6374e

soporteHW, 2017-09-27:
Moved **_sendingATCommands(*)_** function from _southbound_mifare_ file to _southbound_generic_ file.