Sinapse-Energia/MIFARE-US #9: [Basis Code] To upload the basis code generated by CubeMX for Atollic

ralcaide, 2017-07-28:
The basis code should be uploaded. The basis should have the basic configuration and has been generated by CubeMX from Atollic

francisjjp, 2017-07-31:
This project has following modules:

* USART1 for communicating with WIFI module.
* SCL1-SDA1 for communicating with LCD and RFID (both modules are I2C devices).
* PB9 or output pin for enabling/disabling buzzer.

It is needed to define what pin is going to be used in order to switch on/off the main LED.

ralcaide, 2017-07-31:
Several things before close this issue:

- To order the code. All the related docs / code related with CubeMX should be saved in a folder named CubeMX. Please create this folder and move the code to it.
- The file *.elf.launch is an executable, isn't?. If yes, it should be removed from the repository

Also, the design is not end because it is necessary to define the PIN to be used for the main LED. To create another issue to make this definition, and then close this one.

francisjjp, 2017-08-01:
[ARRANGING PROJECT. 0.5 hours] .Closing.

closed 2017-08-01 by francisjjp