Sinapse-Energia/MIFARE-US #10: [Basis Code] To define main LED pin to show status.

francisjjp, 2017-08-01:
It is needed to define the pin to use for showing general status of device.

The lines must be included into MX_GPIO_Init() function. Take account of enabling or disabling some new clock (maybe it is needed to include some __HAL_RCC_GPIOX_CLK_ENABLE() function).

When this issue is done, basic code will be completed.

soporteHW, 2017-08-14:
We'll use PIN A15 for LED_STATUS (the same pin that GPS_LED in Torrot project).

Committed and pushed Code.

ralcaide, 2017-08-14:
Not close the issues until it is tested and the code reviewed. I will close the issues when it is needed.
It is necessary only to move the issues to "DONE" here: https://github.com/Sinapse-Energia/MIFARE-US/projects/1