Sinapse-Energia/MIFARE-US #11: [SB]. Southbound functions to implement.

francisjjp, 2017-08-01:
For this project, following basic SB functions must be implemented into file **southbound_mifare.h** and **southbound_mifare.c** files.

**Stack I2C Functions**

- **MIC-I2C_Send_Data()**. It will be a wrapper for HAL_I2C_Master_Transmit()  function.
- **MIC_I2C_Get_Data()**. It will be a wrapper for HAL_I2C_Master_Receive_IT() Interrupt function. The uses of this function do necessary to implement IRQ routine code associated to I2C reception, it implies to manage RX I2C buffer in similar way to UART.

- RFID_Read_Memory_Block(). This function will use MIC_I2C_Get_Data() function with the I2C RFID appropriate.

- LCD_Write_String(). This function will use MIC_I2C_Send_Data() function with the I2C LCD address.

- LCD_Set_Parameters(). (If needed). This function will use MIC_I2C_Send_Data() function with the I2C LCD address.

**Stack UART functions**

- **MIC_UART_Send_Data()**. It will be a wrapper for HAL_UART_Transmit() function.
- **MIC_UART_Get_Data()**. It will be a wrapper for HAL_UART_Receive_It() interrupt funcion. The uses of this function do necessary to implement IRQ routine code associated to UART recection, it implies to manage RX UART buffer.

- MIC_UART_Send_Data() and MIC_UART_Get_Data() will be used by following functions:

1.  HK_Connect()
2. HK_Disconnect()
3. HK_Set_Connection_Settings()
4. HK_Set_Config()
5 HK_Send_Data(). Basic abstract function to send all data.
6. HK_Get_Data(). Bbasic abstract function to get all data.
7. HK_Get_Signal_Strenght().
8. HK_Set_Working_Mode().

Finally following higher layer wrappers must be created. (One #define _HILINK_DRIVER must be used).

- TCP_IP_Connect will wrap HK_Connect() function. 
- TCP_IP_Disconnect will wrap HK_Disconnect() function.
- TCP_IP_Set_Connection_Settings will wrap HK_Set_Connection_Settings() function.
- TCP_IP_Set_Config will wrap HK_Set_Config() function.
- TCP_IP_Send_Data will wrap HK_Send_Data() function.
- TCP_IP_Get_Data will wrap HK_Get_Data() function.

soporteHW, 2017-08-17:
Created Southbound folder and included soutbound_mifare.h and southbound_mifare.c files. 

Included in this files the LCD_Write_String(); function.

soporteHW, 2017-08-17:
To test the buzzer with "MIC_Set_Digital_Output_status()" function, it has been detected that HAL_GPIO_WritePin() function doesn't work. So, it's necessary to use PWM-signal generator.

soporteHW, 2017-08-18:
Included some functions:

_void **LCD_Display_Update**(void);
void **LCD_SetCursor**(uint8_t x, uint8_t y);
void **LCD_Write_mifare_info**(Device_Status status);
void **LCD_Write_String**(char *string);_

soporteHW, 2017-08-21:
Included in southbound_generic (.c, .h) files the following functions to manage internal Flash Memory:

HAL_StatusTypeDef **_FlashNVM_ErasePage_**(uint32_t page);
HAL_StatusTypeDef **_FlashNVM_Read_**(uint32_t start_address, uint8_t* data_out, uint32_t size);
HAL_StatusTypeDef **_FlashNVM_Write_**(uint32_t start_address, const uint8_t* data_in, uint32_t size);

int **_MIC_Flash_Memory_Write_**(const uint8_t *data_in, uint32_t size);
int **_MIC_Flash_Memory_Read_**(const uint8_t *data_out, uint32_t size);

ralcaide, 2017-09-05:
Explanation Generic vs Specific:

http://sinapse.bitnamiapp.com/redmine/projects/firmware-global/wiki/_Firmware_development_structure#Generic-vs-Specific-functions

soporteHW, 2017-11-21:
Included _**void RTC_Save()**_ function. This function save and updated RTC data in local variable. Also, Context is uploaded.

Commit here: https://github.com/Sinapse-Energia/MIFARE-US/commit/d989031d5a6adbebf8dab22eda546726d048d1dc