Sinapse-Energia/MIFARE-US #12: NTP Syncronization

sinapseenergia, 2017-08-14:
In the initial planning se miss this part.

During inizialization process, NTP have to be setup from server.

This means read time from server and ser into microcontroller RTC. 

This time have to be syncronized each certain time. (1 per day and when reboot is enough)

This os part ir inizialization proccess and afecta to context definition  (NTP server, NTP_server_time..)

soporteHW, 2017-08-29:
It's necessary to include **_MIC_Get_RTC_** and _**MIC_Set_RTC**_ functions.

soporteHW, 2017-09-27:
The following functions was implemented and included in "_southbound_generic_" files.

- uint8_t **_NTP_Sync_**(void);

This function includes all NTP Sinchronization process (Connect to NTP server, send & receive NTP data and translate the received data), i.e, the function calls _TCP_Connect(*)_ and _GET_NTP_Time(*)._ Returns 0 if NTP OK or 1 if NOK.

- void **_Get_NTP_Time_**(unsigned char *buffer);

This function translates the NTP received data, set internal RTC and update the Time_server param in Context.