Sinapse-Energia/MIFARE-US #14: [SB UART Functions - GENERIC] To design & implement the SB functions related to UART

ralcaide, 2017-08-22:
Subpart of #11 

- MIC_UART_Send_Data(). It will be a wrapper for HAL_UART_Transmit() function.

- MIC_UART_Get_Data(). It will be a wrapper for HAL_UART_Receive_It() interrupt funcion. The uses of this function do necessary to implement IRQ routine code associated to UART recection, it implies to manage RX UART buffer.

These functions will be used to communicates with any peripheral that uses the UART. Originally think for the communications features (WIFI & GPRS)

soporteHW, 2017-09-04:
Implemented functions to comunicate with any peripheral by UART interface. This functions are included in "_southbound_generic_" files.

- uint8_t **_MIC_UART_Send_Data_**(UART_HandleTypeDef *huart, unsigned char* messageTX, uint8_t 
               lengthOfmessage, uint32_t timeoutTX);

- uint8_t **_MIC_UART_Get_Data_**(UART_HandleTypeDef *huart, unsigned char* messageRX, uint8_t Size);

soporteHW, 2017-09-04:
Included in "_main.c_" files the following functions to use UART interrupts.

- void **_HAL_UART_RxCpltCallback_**(UART_HandleTypeDef *huart)

- void **_HAL_UART_TxCpltCallback_**(UART_HandleTypeDef *huart1)

- void **_HAL_UART_ErrorCallback_**(UART_HandleTypeDef *huart)

francisjjp, 2017-09-25:
Following functions are been implemented and included into southbound_generic.X files.

`void MIC_SPI_Write( SPI_HandleTypeDef *hspi1,uint8_t value)`

- This function sends one byte by SPI hardware.

`uint8_t MIC_SPI_Read(SPI_HandleTypeDef *hspi1,uint8_t readSize)`

- This function receives one value by SPI hardware.

`uint8_t MIC_SPI_TransmitReceive( SPI_HandleTypeDef *hspi1,uint8_t byte )`

- This function sends and receives one byte by SPI hardware.

Of course it is needed to inialize SPI hardware with this:

(Into main.c we configure SPI hardware)

```

/* SPI1 init function */
static void MX_SPI1_Init(void)
{

	 hspi1.Instance = SPI1;
	  hspi1.Init.Mode = SPI_MODE_MASTER;
	  hspi1.Init.Direction = SPI_DIRECTION_2LINES;
	  hspi1.Init.DataSize = SPI_DATASIZE_8BIT;
	  hspi1.Init.CLKPolarity = SPI_POLARITY_LOW;
	  hspi1.Init.CLKPhase = SPI_PHASE_1EDGE;
	  hspi1.Init.NSS = SPI_NSS_SOFT;
	  hspi1.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_16;
	  hspi1.Init.FirstBit = SPI_FIRSTBIT_MSB;
	  hspi1.Init.TIMode = SPI_TIMODE_DISABLE;
	  hspi1.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
	  hspi1.Init.CRCPolynomial = 7;
	  hspi1.Init.CRCLength = SPI_CRC_LENGTH_DATASIZE;
	  hspi1.Init.NSSPMode = SPI_NSS_PULSE_ENABLE;
	  if (HAL_SPI_Init(&hspi1) != HAL_OK)
	  {
	    _Error_Handler(__FILE__, __LINE__);
	  }

}

```

Into stm32f0xx_hal_msp.c we write following code:
```
void HAL_SPI_MspInit(SPI_HandleTypeDef* hspi)
{

  GPIO_InitTypeDef GPIO_InitStruct;
  if(hspi->Instance==SPI1)
  {
  /* USER CODE BEGIN SPI1_MspInit 0 */

  /* USER CODE END SPI1_MspInit 0 */
    /* Peripheral clock enable */
    __HAL_RCC_SPI1_CLK_ENABLE();

    /**SPI1 GPIO Configuration
    PB3     ------> SPI1_SCK
    PB4     ------> SPI1_MISO
    PB5     ------> SPI1_MOSI
    */

    GPIO_InitStruct.Pin = GPIO_PIN_3|GPIO_PIN_4|GPIO_PIN_5;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF0_SPI1;
    HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

  /* USER CODE BEGIN SPI1_MspInit 1 */

  /* USER CODE END SPI1_MspInit 1 */
  }

}

void HAL_SPI_MspDeInit(SPI_HandleTypeDef* hspi)
{

  if(hspi->Instance==SPI1)
  {
  /* USER CODE BEGIN SPI1_MspDeInit 0 */

  /* USER CODE END SPI1_MspDeInit 0 */
    /* Peripheral clock disable */
    __HAL_RCC_SPI1_CLK_DISABLE();

    /**SPI1 GPIO Configuration
    PB3     ------> SPI1_SCK
    PB4     ------> SPI1_MISO
    PB5     ------> SPI1_MOSI
    */
    HAL_GPIO_DeInit(GPIOB, GPIO_PIN_3|GPIO_PIN_4|GPIO_PIN_5);

  /* USER CODE BEGIN SPI1_MspDeInit 1 */

  /* USER CODE END SPI1_MspDeInit 1 */
  }

}
```

**Clock core ARM is 26MHz, clock for APB1 that drives SPI must be 13MHz.**

Lines uses are:

PB3 for SPI1_SCK
PB4 for SPI1_MISO
PB5 for SPI1_MOSE.

For chip select, CS,  is used: PB13.

Into main.c we write for CS following code... 

 ```
  /*Configure GPIO pin : CSS_Pin */
    GPIO_InitStruct.Pin = CSS_Pin;
    GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH ;
    HAL_GPIO_Init(CSS_GPIO_Port, &GPIO_InitStruct);
```

francisjjp, 2017-09-25:
I suppose I can close issue.