Sinapse-Energia/MIFARE-US #15: [SB RFID Functions - GENERIC & MIFARE] To design & implement the SB functions in order to use the RFID

ralcaide, 2017-08-22:
Subpart of #11 

To design and develop the GENERIC function
- RFID_Read_Memory_Block(). This function will use MIC_I2C_Get_Data() function **or** a library function in order read the RFID

Then should be developed a MIFARE SB function that call the GENERIC one in order to get the RFID data needed for the MIFARE project.

soporteHW, 2017-09-24:
Now, it's necessary to implement **_RFID_Read_Memory_Block()_** function to wraps allf process: Check+Anticoll+SelectTag+Authentication+Reading.

francisjjp, 2017-09-25:
The function is:

`int RFID_Read_Memory_Block(int blockTrail, int blockRead, unsigned char *buffer);`

It is needed to include two blocks: BlockTrail for authentication and BlockRead for reading the block . In our example is RFID_Read_Memory_Block(63,62, bufferRFID);

francisjjp, 2017-09-25:
I suppose I can close.

soporteHW, 2017-09-27:
Moved **_RFID_Read_Memory_Block_**(*) function from _tm_stm32f4_mfrc522_ file to _southbound_generic_ file.