Sinapse-Energia/MIFARE-US #16: [SB LCD Functions - GENERIC & MIFARE] Design & Develop the SB functions in order to use the LCD

ralcaide, 2017-08-22:
Subpart of #11 

To develop the following GENERIC SB functions

- LCD_Write_String(). This function will use MIC_I2C_Send_Data() function with the I2C LCD address **or** a LCD library function .
- LCD_Set_Parameters(). (If needed). This function will use MIC_I2C_Send_Data() function with the I2C LCD address **or** a LCD library function.
- LCD_* : All the functions needed to work with the LCD in a generic way

The above functions should be generic and will be called by the SB MIFARE functions in order to use the LCD as needed by MIFARE. The LCD MIFARE SB functions should be also developed through this issue

ralcaide, 2017-08-23:
Please, put here the links to the commits where the job was developed

soporteHW, 2017-08-23:
Job was developed in https://github.com/Sinapse-Energia/MIFARE-US/commit/e5375032defaf934de5d9cf1f5aa54666588b444

francisjjp, 2017-09-25:
I suppose I can close.

soporteHW, 2017-09-27:
Modifications in LCD_Write_String(*) function. New definition is:

_void **LCD_Write_String**(char *string, FontDef sizefont);_

Now, we can select the font size to display on LCD (options: 7x10 ; 11x18, 16x26 defined in ssd1306 library).