Sinapse-Energia/MIFARE-US #17: [SB Communications Functions - GENERIC] To design & develop the SB functions in order to communicates using HiLink as a WIFI driver

ralcaide, 2017-08-22:
subpart of #11 

The following functions should be developed in order to use the HiLink device to as a TCP_IP transceiver

- HK_Connect()
- HK_Disconnect()
- HK_Set_Connection_Settings()
- HK_Set_Config()
- HK_Send_Data(). Basic abstract function to send all data.
- HK_Get_Data(). Bbasic abstract function to get all data.
- HK_Get_Signal_Strenght().
- HK_Set_Working_Mode().

Finally following higher layer wrappers must be created. (One #define _HILINK_DRIVER must be used).

Then, the TCP_IP functions should be also created wrapping the HK_* functions if _HILINK_DIRVER is defined. The TCP_IP functions wraps all the possible methods of connection. Here will wrap only HILINK

- TCP_IP_Connect will wrap HK_Connect() function.
- TCP_IP_Disconnect will wrap HK_Disconnect() function.
- TCP_IP_Set_Connection_Settings will wrap HK_Set_Connection_Settings() function.
- TCP_IP_Set_Config will wrap HK_Set_Config() function.
- TCP_IP_Send_Data will wrap HK_Send_Data() function.
- TCP_IP_Get_Data will wrap HK_Get_Data() function.

These functions will be used by the NB (HTTP Client) in order to send & receive HTTP messages

ralcaide, 2017-08-24:
The HTTP client will be only a wrapper of the TCP_IP functions in order to:
- Connect to the server
- Send a request
- Receive response

soporteHW, 2017-09-04:
Implemented the following function included in "southbound_mifare" file:

- HLKStatus _**HK_Set_Config**_ (UART_HandleTypeDef *phuart1, uint32_t retries,
		uint32_t timeoutTx, uint32_t timeoutRx, unsigned char *messageRX);

This function configure HLK driver parameter's via _AT commands_. By default, HLK driver parameter's will be configured via _Web Server_ interface.

ralcaide, 2017-09-05:
Comments added to the commit. Please take them into account

soporteHW, 2017-09-05:
Changes related to commit https://github.com/Sinapse-Energia/MIFARE-US/commit/499e0853622cacccb65443efc3f5df0c7ad6374e

- Moved **_HK_*_**() functions from _southbound_mifare_ file to _southbound_generic_ file.
- Some modifications in **_HK_Set_Config_**(*) function. Now, this function only implements the AT commands to configure UART and Networks parameters.
- Implemented **_HK_Get_Config_**(*) function. This function get the internal configuration parameters. In our case, we'll use it to get IP_device and IP_server in order to save them into Context.
- Implemented **_HK_Connect_**(*) function. This function set Socket connection parameter's (IP_server and Port) via AT commands.

soporteHW, 2017-09-20:
Implemented the followings functions:

- TCPStatus **_TCP_Connect_**(HK_Working_Mode mode, HK_Network_Mode netmode, UART_HandleTypeDef *phuart, uint32_t retries, uint32_t timeoutTx, uint32_t timeoutRx, unsigned char *messageRX);

This functions returns 0 if Connect OK or 0 Connect NOK.

- uint8_t **_NTP_Sync_**(uint8_t NTP_Sync_state);

This function includes all NTP Sinchronization process (Connect to NTP server, send & receive NTP data and translate the received data).

- void **_Get_NTP_Time_**(unsigned char *buffer);

This function translates the NTP received data, set internal RTC and update the Time_server param in Context.

soporteHW, 2017-09-20:
Implemented the following functions:

- TCPStatus **_TCP_Set_Config_**(HK_Working_Mode mode, UART_HandleTypeDef *phuart, uint32_t retries, uint32_t timeoutTx, uint32_t timeoutRx, unsigned char *messageRX);

This function calls _HK_Set_Config(*)_ in order to configure the device network parameters. Returns 0 if Set OK or 1 if NOK.

- TCPStatus **_TCP_Get_Config_**(HK_Working_Mode mode, UART_HandleTypeDef *phuart, uint32_t retries, uint32_t timeoutTx,uint32_t timeoutRx, unsigned char *messageRX);

This function calls _HK_Get_Config(*)_ in order to Get the device network parameters. For now, ths function only get IP_device and IP_server parameters, the most important parameters for our application. This parameters are saved in Flash and Context structure.