Sinapse-Energia/MIFARE-US #18: [SB LED&BUZZER - GENERIC & MIFARE] Design & implement SB functions related with LED and BUZZER devices

ralcaide, 2017-08-22:
Subpart of #11 

For controlling LED and BUZZER following generic function will be used.

- MIC_Set_Digital_Output_status(). This function put 1 or 0 over one pin in one defined port. This function wraps HAL_GPIO_WritePin() function. It is supposed that is not needed to use oscillator/PWM signal in order to do work buzzer. If not, one PWM-signal generator function should be used.

The above functions are generic and can be used in any project. Maybe it is necessary to create some MIFARE SB functions in order to ease to provide to concrete behavior required in this project

soporteHW, 2017-08-23:
Implementes _void **MIC_Set_Digital_Output_status**(GPIO_Pin_Select pin, PIN_Status status);_ function.

This function receives as parameter the GPIO_Pin to use and the status to write (ON/OFF)

francisjjp, 2017-09-25:
I suppose I can close.