Sinapse-Energia/MIFARE-US #19: [SB NVM - GENERIC] Design & Implement the SB functions that ease the use of a NVM

ralcaide, 2017-08-22:
Subpart of #11 

It is necessary to save the Context variables in a NVM, so it is necessary to ease the management of this NVM through several generic SB functions. 
In this project, we will use a part of the flash as a NVM, it should be developed the following GENERIC SB functions

- HAL_StatusTypeDef FlashNVM_ErasePage(uint32_t page);
- HAL_StatusTypeDef FlashNVM_Read(uint32_t start_address, uint8_t* data_out, uint32_t size);
- HAL_StatusTypeDef FlashNVM_Write(uint32_t start_address, const uint8_t* data_in, uint32_t size);

- int MIC_Flash_Memory_Write(const uint8_t *data_in, uint32_t size);
- int MIC_Flash_Memory_Read(const uint8_t *data_out, uint32_t size);

soporteHW, 2017-08-23:
Developed all functions defined above.

Moreover, _**MIC_Flash_Memory_Write/Read**_ uses a data struct to write/read in Flash, so it's necessary create this data struct with Context variables. Also, it will be necessary create a function to copy Context variables to RAM variables

ralcaide, 2017-08-23:
@soporteHW Please, put here the links to the commits where the job was developed

soporteHW, 2017-08-23:
Job was developed in https://github.com/Sinapse-Energia/MIFARE-US/commit/4ce405e60af65810c644b74e7790a7567a3b1c72

francisjjp, 2017-09-25:
I suppose I can close.