Sinapse-Energia/MIFARE-US #20: [BUG] Problems with HK TCP Client connection

soporteHW, 2017-09-07:
Tested that HK_Set_Config(*) and HK_Connect(*) works fine. 

Wifi gateway was configurated as follow:

[image]

[image]

WIFI connects with IP: 192.168.1.164 (PC IP address). With Wireshark app capturing paquets,  If we send an HTTP message like this "_MIC_UART_Send_Data(&huart1,(uint8_t*)"GET / HTTP/1.1\r\nHost: 192.168.1.164\r\n\r\n",36,100);_" , the message is sent OK.

[image]

I'm trying to simulate am HTTP local server with http://miniweb.sourceforge.net/ in order to validate the TCP connection. It seems works OK, server detects the wifi client connected.
[image]

So, when I send the same HTTP message, it doesn't work.

ralcaide, 2017-09-08:
The server is running in your local in the port 8000, isn,t? I think your HTTP message should include this port but I'm not sure because is already included in the gateway configuration. Anyway, try it.

Test also with other HTTP Client trying to perform a request to your server. The google chrome app HTTP Rest Client is very useful to test these kind of things

soporteHW, 2017-09-08:
Modified the HTTP message format:

 _MIC_UART_Send_Data(&huart1,(uint8_t*)"GET /index.htm HTTP/1.1\r\nHost: 192.168.1.164\r\n\r\n",48,100);_ 

[image]

Now, HTTP message are sent OK.

soporteHW, 2017-09-08:
We receive valid HTTP response with using one example HTTP local server.

As shown in figure, we receive 2383 bytes from HTTP server.

[image]

closed 2017-09-11 by francisjjp