Sinapse-Energia/MIFARE-US #21: [BUG]. General problems with HK wifi-ethernet module

francisjjp, 2017-09-11:
There is a big problem with HK devices.

After some time. Ethernet port blinks both LEDs and it is not possible to configure anything to HK device. Neither using AT commands, nor using wifi mode.

This issue has been reported to HILINK company.

soporteHW, 2017-09-14:
As show in datasheet, to restore device factory settings it's necessary to follow this steps:

"ES0 Function:
1. After the system starts properly, ES0 left low 0.05s <t <6s, Serial port 0 to exit transparent
transmission mode.
2. After the system starts properly, ES0 left low 6s <t <12s, restore customer settings. * See
"parameters are stored."
3. ES0 low power, low time 6s <t <12s, restore customer settings. * See "parameters are stored."
ES1 Features:
1. After the system starts properly, ES1 left low 0.05s <t <6s, Serial port 1 to exit transparent
transmission mode.
2. After the system starts properly, ES1 left low 6s <t <12s, restore factory settings. * See
"parameters are stored."
3 Quick Start
3.1 Restore factory settings
In order to ensure that all of the configuration process is correct，bring the module to restore the
factory settings first. If it is already in factory mode, you can skip this step. Power on the module
with 5V (350mA) electric source, wait about 30 seconds, After starting, Lower ES/RST pin more
than Trst, release of ES/RST feet, the system will automatically restart. After rebooting the system
the module is already in Factory mode."

We have program two m2m device:
1) With FW that put low ES1 pin during 10s.
2) With FW  that pu low ES0 pin during 10s.

If we connect the IoTHub in oscilloscope, it show the correct sequence but the RM08S doesn't restore.
Also, we have tried to power on the RM08S with a external power supply (checking if problem was de DC-DC). The results are NOK.

soporteHW, 2017-09-15:
Devices keep failling (today 3 more devices have failed).

The following test have checkd:

- One failed RM08S device has soldered pint to pin in the EVB kit, in order to try restore factory settings pressing restore button. The results were NOK, the device restarts all time and it doesn't properly start. Possible errors: DC-DC phase, noise through Ethernet interface, shorcircuit???
- A new model RM08M device has been soldered in IoTHub base. This device has equivalent pinout than RM08S module. At the moment, it works fine.
For test, three devices have been connected: 2 devices with RM08S wifi model (without m2m module, power suplly USB 5V-1A) and 1 RM08M device wifi model (with m2m module connected too, power supply USB 5V-2A). *_See the attached images_

NOTE: in one RM08S device, the Ethernet interface does not seem to work properly.

[image]
[image]

soporteHW, 2017-09-18:
At date 18/09/2017 the devices status is as follows:

- ID_13.4: WIFI interface = OK; Ethernet interface = OK;
- ID_3.2: WIFI interface = OK; Ethernet interface = NOK;
- ID_13.3: WIFI interface = OK; Ethernet interface = OK; m2m interface = OK;

Note: In device ID_13.3 ,the internal RTC has accumulated an offset (one hour delayed).