Sinapse-Energia/MIFARE-US #22: [TEST] Http transimition and XML processing

sinapseenergia, 2017-09-20:
This is an END to END test to be performed.

Stepts - Set tests enviroment
//Actual message formats can be checked from VPN Linux terminal available at office local network
//As easy as send POST message from terminal to server, and analyze message response format

Conect a linux terminal with an IP visible from Device Network Config. (For instance a Raspberry)
* Send XML via HTTP from device to Linux terminal. Check if received as expected (Actual format)
* Send XML response via HTTP from Linux Terminal to device. Check if received data is correctly interpreted and stored

ralcaide, 2017-09-20:
@sinapseenergia , what should I do related to this issue?

sinapseenergia, 2017-09-20:
@ralcaide , tech support if needed

soporteHW, 2017-09-21:
Trying to simulate HTTP transmission using Raspberry as server and sending GET request via WIFI device. Results are NOK.

[image]

soporteHW, 2017-10-05:
Some tests have been performed in local server (using Wireshark and Node-Red application) with successful results. 

Today 05/10/2017 two devices have been installed in Domingo's office to check the device operation. Previously, Raul did some test with successful results in both devices.