Sinapse-Energia/MIFARE-US #28: [BUG] Device freezes or RFID doesn´t read anymore

sinapseenergia, 2017-10-19:
After a while connected , device Freezes

We can not know if program is running due to the LCD just Show "TESTING"

Probably program is runnig, due to LCD write is a bucle, and WDT is enabled

In this case, under unknow circumpstances, devices doesn´t read RFID anymore

Debug and fix this fail

sinapseenergia, 2017-10-26:
It seems a problem with program flow.

Under circumpstances, if suddently Ethernet falls (Or unplugged). Device doesn´t work anymore.

Desired behaviour:

If not server response, Show message indicating (And Buzzer)

Go back to normal behaviour.

If last 10 attemps to connect to server fails (consecutive, 1 OK reboot counter) , show message in LCD. "Servidor fuera de linea"

This is just an informative message, under this situation, when RFID pass again, message anyway is sent

soporteHW, 2017-11-21:
Included WatchDog funcionality. System restart if module freeze.